analogdevicesinc/scopy #88: Connecting to a device

DanielGuramulta, 2017-04-26:
When clicking on the add button, a new page should be opened where we have the following

[image]
The connect button will search for the device with the given ip, Add will work only when we have found a device, and on click it will add the device, abort will return to the normal page that is usually displayed.

[image]

pcercuei, 2017-04-26:
So how would that be different from what we already have?

DanielGuramulta, 2017-04-26:
We would be able to use the application , while it is searching for a device.

pcercuei, 2017-04-26:
I don't really understand the point, there is nothing else that the user would or could do than waiting while trying to connect to a device.

dNechita, 2017-04-26:
I think there are some enhancements that this approach would bring:
- avoids the application to hang (both the window and the client area). Even if the hang takes a couple seconds it's never a good idea to have things hanging, especially the window that currently cannot be moved, minimized, etc.
- avoids using dialogs. This way it gets to be consistent with the rest of the application.
- this might be the starting point of a more complex device manager.  A device manager could be used to interrogate the device (calibration details, temperature, hardware revision, firmware revision, etc), remove it from the list of devices, perform a custom calibration and other actions that we might add in the future.

amiclaus, 2018-05-11:
On Scopy build: 07076df (v0.12) the feature is implemented.

closed 2018-05-11 by amiclaus