analogdevicesinc/scopy #89: Export/Import Waveform

amiclaus, 2017-05-03:
The export/import waveform features should be implemented for the Oscilloscope, since they are required for the ADALM2000 Based Lab Activity Material -  [https://wiki.analog.com/university/courses/electronics/electronics-lab-1st](Op Amp Settling Time) (see Procedure).

rgetz, 2017-05-03:
to be clear - there isn't any such concept as import to the oscilloscope - is there?

exports are defined in #3 
imports are defined in #37 

so, this is a dup?

amiclaus, 2017-05-03:
1. I've corrected the wording of the issue.

2.Yes, it is a duplicate. From what I can see, the issues are pretty old and still not solved. 
I tagged it as a high priority since these features are used in one of the first ADALM2000 lab activities.

damercer, 2017-05-04:
There may actually be one or two cases where loading a "waveform" to be displayed on the Scope screen would be useful. First, Reloading a saved / exported waveform from previous session to compare it to current waveform. Second, loading a waveform from simulation to compare it to the current waveform.

DM

rgetz, 2017-05-08:
Sure, import a waveform to the input analog channels is known as a "loading a reference waveform".

http://www.tek.com/manual-topic/using-reference-waveforms-and-traces

http://www.tek.com/manual-topic/saving-and-recalling-waveform-and-trace-data

https://www.youtube.com/watch?v=0TqgIMZh-Wo

http://literature.cdn.keysight.com/litweb/pdf/75019-97073.pdf (chapter 5)

-Robin

adisuciu, 2018-03-07:
Implemented with https://github.com/analogdevicesinc/scopy/pull/379

amiclaus, 2018-03-12:
On Scopy build: 6bd7075 (v0.11) the feature is implemented.

closed 2018-03-12 by amiclaus