analogdevicesinc/scopy #90: Connecting over USB

DanielGuramulta, 2017-05-04:
When we plug a new usb device ,if scopy is running it will not appear in the device list until we restart the application.
Expected Result: The device list should update automatically when a usb device is connected.

dNechita, 2017-05-04:
@pcercuei
Does (or will) libiio have some kind of hotplug event notification to detect new/removed usb contexts?

pcercuei, 2017-05-04:
We thought about that at some point, but we never went to actually implement it in libiio.

For now, there could be a thread in Scopy that constantly scans the USB devices (there's a libiio API for that) every 5 seconds or so, and automatically add the new ones found (and delete the ones that have been removed).

DanielGuramulta, 2017-05-11:
I added a commit that implements this enhancement using a thread to scan the USB devices: <tt>2e6a2169d14f2d05b1cdd2c17593efbff8a43ec6</tt>

pcercuei, 2017-05-11:
/home/paul/dev/ADI/scopy/src/tool_launcher.cpp:138:47: error: invalid use of non-static member function ‘QPair<iio_context_info**, int> adiscope::ToolLauncher::searchDevices()’

dNechita, 2017-05-11:
This https://github.com/analogdevicesinc/scopy/commit/7c76dbab03ff6cd04091e49e47f49969964f0899 should solve the build error.

pcercuei, 2017-05-11:
That works better, yes. Thanks for the fix.

closed 2017-05-11 by pcercuei