analogdevicesinc/scopy #92: Calibration correction: Apply in hardware + HDL + firmware or in software clients?

dNechita, 2017-05-09:
Once a calibration is done and offset and gain errors are found there is a decision to be made on where the corrections should be made.
1. Applying all calibration correction in a client software (like Scopy) will mean that no other M2K software clients will benefit from these corrections. However calibration results could be stored by the device and read by any software. Doing corrections in software means better approximations due to using floating point values.

2. On the other hand correcting things in hardware allows every software client to have access to samples that are already calibrated. But precision might be affected.
The samples that are sent/received by the M2K are integer which means additional errors will be added. (E.g. trying to correct an offset of 1.5 LSB, the error would be 0.5 LSB)

acostina, 2017-06-14:
Added scale correction as part of the IQ module and activated it as part of the m2k design. The offset is compensated by an outside DAC. If needed, the DC filter module can be activated as part of the chain, but at this point it's not needed.

lclausen-adi, 2017-06-19:
Driver support: https://github.com/analogdevicesinc/linux/commit/098d8b76163c8dfc1341992ec5afacbf52e66966

adisuciu, 2020-10-23:
Calibration is now entirely stored on the device.

https://github.com/analogdevicesinc/libm2k/pull/120
https://github.com/analogdevicesinc/libm2k/pull/132
https://github.com/analogdevicesinc/libm2k/pull/158

 There is also the possibility to do calibraiton based on a look-up table with temperature. We will implement this option in Scopy as well.

https://github.com/analogdevicesinc/libm2k/pull/151

closed 2024-09-24 by adisuciu