analogdevicesinc/scopy #94: Signal Generator: waveform settings

DanielGuramulta, 2017-05-12:
When selecting the mHz option for the frequency in the waveform tab:
[image]
Scopy will hang for about 50-60 seconds before changing it;
[image]
Also if we use the signal generator with the mHz option and then disconnect the device, when connecting it again, scopy will hang for the same amount of time before we can use its tools:
[image]

pcercuei, 2017-05-12:
Yes, I don't know why, GNU Radio chokes when generating slow waveforms...

msuciu, 2017-07-07:
If it helps I receive this error.
[image]

adisuciu, 2017-10-20:
This should be fixed with preview window optimizations - f2876e55e1d1ece4a1b2f3d358a623d9a0368e55 and f3e3f436ce07e6fc18038d2370f5f0a8163951df

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the bug is fixed.

closed 2017-11-21 by amiclaus