analogdevicesinc/scopy #95: Signal Generator: Phase of Sine vs Triangle

damercer, 2017-05-14:
With the phase set to 0 the sine wave shape starts at zero ( mid point of wave ) and increases with time. With the phase set to 0 the triangle wave shape starts at the maximum ( positive peak ) value and decreases. These are obviously different from each other, 270 degrees out of phase ( or -90 ) based on rising zero crossing. I would think they should both be the same. i.e. for 0 phase start at the rising "zero crossing" ( mid value ).

Doug

damercer, 2017-05-14:
BTW the square wave shape appears to be 180 degrees out from the sine but the rising / falling ramps seem to be "correct".

Doug

pcercuei, 2017-05-15:
Should behave properly now, thanks for finding this.

commit a5ce5009 (2017-05-15): Signal Generator: Fix phase of triangle and square waveforms

Fixes #95.
closed 2017-05-15 by pcercuei