analogdevicesinc/scopy #96: Network Analyzer: Frequency scale should be log

damercer, 2017-05-16:
Hi:
First a general question. Is the network analyzer in a working state? It seems to not plot anything on the graph and go away and never come back. Stop button does nothing.

Second. the frequency axis should be log scale ( optionally linear ). Frequency steps should be in number of points per decade so as to be evenly spaced on log scale.

Doug

pcercuei, 2017-05-16:
Hi Doug,

The network analyzer is in a working state in the 'network_analyzer' branch - you can try this build: https://ci.appveyor.com/project/analogdevicesinc/scopy/build/1.0.502/artifacts

Why should the frequency be log scale? (genuine question)

damercer, 2017-05-16:
The span of frequencies can be very large i.e multiple decades so it is 
easier to see fine details on log scale. It is also tradition. Bode 
plots are almost always log / log plots for amplitude ( gain in dB ) vs 
frequency ( log ). The slope of a frequency response is most often given 
in so many dB/decade ( or octave ) so on a log scale this will be a 
linear line. i.e first order slope is 20 dB/decade.

Doug

damercer, 2017-05-25:
This should be bumped up to a Requirement rather than enhancement. By definition a Bode frequency plot has a log horizontal axis:

From: https://en.wikipedia.org/wiki/Bode_plot

In electrical engineering and control theory, a Bode plot /ˈboʊdi/ is a graph of the frequency response of a system. It is usually a combination of a Bode magnitude plot, expressing the magnitude (usually in decibels) of the frequency response, and a Bode phase plot, expressing the phase shift. Both quantities are plotted against a horizontal axis proportional to the **logarithm of frequency**. Given that the decibel is itself a logarithmic scale, the Bode amplitude plot is log–log plot, whereas the Bode phase plot is a lin-log plot.

Doug

pcercuei, 2017-05-30:
@rgetz, do we want the possibility to use a linear scale as well?
Implementing both will probably be a pain, so I'd prefer doing only logarithmic.

damercer, 2017-05-30:
Just my 2 cents.... There are a few cases where a linear scale is appropriate. This goes manly for the spectrum analyzer but but for the network analyzer as well. When the displayed frequency range is very small, roomed into a limited bandwidth, the log scale is not needed and having things spaced linearly is the best way to view the measurements.

I don't know how hard things are to do in the tools you are using but in my experience writing ALICE in Python doing both lin and log scales was trivial.

Doug

commit a9accb35 (2017-06-01): NetworkAnalyzer: Add support for logarithmic sweep

Fixes #96.
closed 2017-06-01 by pcercuei