analogdevicesinc/scopy #97: Tool_launcher: connect button

DanielGuramulta, 2017-05-16:
If we select a device in scopy: 
[image]
and then unplug it, scopy will remove it from the device list but the connect button will remain enabled:
[image]
Clicking it in this situation will make scopy crash:
[image]

DanielGuramulta, 2017-05-16:
This commit <tt>https://github.com/analogdevicesinc/scopy/commit/5309373a58e40745082216e02ccfe6a3ad518c29</tt> should solve the issue.

pcercuei, 2017-05-17:
The commit message isn't very informative...
You could write something like "Tool Launcher: Disable Connect button if no devices selected"
and that would already greatly help for understanding what the commit does.

DanielGuramulta, 2017-05-17:
I have edited the commit: <tt>https://github.com/analogdevicesinc/scopy/commit/d92235cd7cd4ac469c07fbe3fd882d4c341f8244</tt>

pcercuei, 2017-05-17:
Ok, I pushed your commit to master. Thanks for the bugfix!

DanielGuramulta, 2017-05-17:
Due to the changes of the disconnect() method in src/tool_launcher.cpp, the fact that it also updates the current list of devices, caused a crash when trying to delete a device that was selected but no longer connected via USB. When calling the click() method on the button:
`(*it)->second.btn->click();` 
the method disconnect() will be called in the device_btn_clicked() method and then the updateListOfDevices 
will be called again(the first time it is called after the watcher emits finished()) `updateListOfDevices(searchDevices());` 
Being called twice one gets to delete the device and the other crashes when reaching the line:
`QString uri = (*it)->second.btn->property("uri").toString();`
This new commit solves the new bug: <tt>https://github.com/analogdevicesinc/scopy/commit/567d1ed6d446e3ca6404bd5f2c8622d2ef9e13b4</tt>

pcercuei, 2017-05-17:
Feel free to push it to master.

If you add "Fixes #97" to the commit message, this bug will be automatically closed with a reference to the commit that fixed it.

closed 2017-05-18 by DanielGuramulta