analogdevicesinc/scopy #98: Network Analyzer: Choosing the active channel and the reference one

dNechita, 2017-05-17:
Since there are two configurations available:
Channel 1 (of oscilloscope and signal generator) - the active network analyzer channel
Channel 2 - the reference channel
or
Channel 1  - the reference channel
Channel 2 - the active network analyzer channel
(which is currently implemented)

Should we provide users the option to easily switch between these 2 configurations?

rgetz, 2017-05-17:
I don't think it is nessary. Channel one can always be the reference, and channel 2 can always be the network analyzer Chanel.

damercer, 2017-05-17:
Robin is correct you can always move the wires but it would be nice to just swap the channels in software when you want the reverse transfer function.

Doug

amiclaus, 2017-05-30:
Currently channel 2 is the reference, according to Scopy.

damercer, 2017-05-30:
This backward to the way Digilent Waveforms does it ( I think we are trying to make it simple for professors to make the transition ). The source is assumed to be connected to channel 1 and is thus the reference and the output is on channel 2. Also there are two modes, One uses Channel 1 as the reference and the displayed trace is the difference in dB ( ratio in volts ) or gain from "input" to "output". The other mode does not consider channel 1 as a reference level and just displays the channels in absolute dB.

Doug

commit 78670493 (2017-06-05): NetworkAnalyzer: Add option to select the reference channel

Fixes #98.
closed 2017-06-05 by pcercuei