analogdevicesinc/scopy #99: Switching Channel Selection generates Scopy crash

amiclaus, 2017-05-18:
Connect M2K. Start Scopy. Select Oscilloscope feature. Run Oscilloscope. Start switching channel selection. Scopy tool stops responding.
[image]

It is to be mentioned that the previous situation is not valid if the Scopy tool is opened, and after calibration is restarted.

pcercuei, 2017-05-18:
I can't reproduce...

rgetz, 2017-05-18:
Antoniu:

What git tag of Scopy? What operating system? What firmware?

pcercuei, 2017-05-18:
@rgetz I think it's been fixed already, about one hour before @amiclaus reported the bug.
I'm waiting for his confirmation and then the bug report can be closed.

amiclaus, 2017-05-18:
Fixed via [60c2d62](https://github.com/analogdevicesinc/scopy/commit/60c2d622e74334ec554d65da78d8d983e94fb05c) . My Scopy version was two commits behind (Took the build in the morning).

closed 2017-05-18 by amiclaus