analogdevicesinc/scopy #100: O-scope, Measurments display: Not using screen space efficiently

damercer, 2017-05-19:
Hi:
I have a few atheistic comments about how the measurements are displayed. When the user is custom selecting what is displayed the items are added in vertical columns starting on the left above the waveform grid. As we see here:
[image]
If only the four items are displayed as we see, there is a large blank area left over. This shrinks the waveform display grid to a much smaller region of the overall window than is necessary. It would be a better ( more efficient ) use of screen real estate if the items were added in horizontal rows rather than vertical columns.

Also, as we can see, the items are placed in the order they were added. It would be more logical if they were grouped by channel. And within a channel's grouping by vertical or horizontal measurement type. This will make it easy for the user when looking through the
list. Color coding the text by channel is great but grouping them in addition would be even better.

One more suggestion related to how the items are arranged. If they were arrayed in horizontal rows it might be good if channel 1 stuff used the top row(s) first and then channel 2 stuff started on the next new row under the channel 1 stuff. 

Doug

rgetz, 2017-05-19:
if you click on the "scopy" icon, the instrument names should collapse. (not what you asked for, but will make things larger); but yes - I agree with what you are saying.

-Robin

damercer, 2017-05-19:
Right, that helps some but it makes the grid long and skinny.

I think the general mind set when designing the display window(s) is to always be thinking about how to maximize the space used by the grid. That is after all what the user wants to concentrate on.

Doug

acozma, 2017-05-22:
I'm working with the GUI designer to provide alternative ways to display the measurements. Will come back shortly with some options.

rgetz, 2020-01-08:
Just wanted to pop this back up to the top (sort of).

https://wiki.analog.com/university/tools/m2k/scopy/oscilloscope#using_built-in_signal_measurement_features

when the screen is smaller than 1080p, all the measurements aren't displayed?

adisuciu, 2020-01-09:
No, the issue refers to the measurements being displayed vertically instead of horizontally, wasting some space (in the picture provided).
We implemented a scroll bar a while ago, which is visible in the link you provided (on top)