analogdevicesinc/scopy #103: Improving Scopy interactivity

damercer, 2017-05-20:
This request it aimed at making the controls for the Scopy instruments more interactive ( i.e. faster to use and less annoying ). Right now if the user has say a series RLC circuit set up for testing and wants to see how the voltage across the inductor ( or capacitor ) changes as the frequency and amplitude of the square wave from the signal generator is adjusted he has to constantly switch back and forth between the scope and generator screens. While changing the generator he can't see the waveforms change and visa versa when he is observing the waveforms he can't change the generator. As you can see this situation can become very annoying. This is also true when adjusting the DC user supplies for example.

My request; is it possible to include a minimal set of signal generator ( and DC power supply ) controls that slide on and off the right hand side of scope display like the various scope controls do?

The optimal solution would be to have individual windows for each instrument but this suggestion is at least a partial answer that may fit in the current design.

Thanks

Doug

acozma, 2017-05-22:
The plan is to have individual windows for each instrument. This was specified in the original design but has not made it into the implementation yet.

damercer, 2017-05-22:
That will be great. I know there is a lot still needing to be done to Scopy but any idea where this might be on the priority list? I would think users would like to see it sooner rather than later.

Separate windows is a major departure from the way it is now. Again maybe best to get it done before doing work that will need to be un-done or re-done?

Is this a goal to have the separate windows in the first production release?

Thanks

Doug

pcercuei, 2017-05-22:
It shouldn't be too hard to do. I can have a look at it this week.

dNechita, 2017-05-25:
@acozma Can you request a mockup for the detach button that we will need to add to each instrument?

DanielGuramulta, 2017-09-06:
In order to detach a tool, just drag it from the left menu and drop it somewhere else. (Available since commit: https://github.com/analogdevicesinc/scopy/commit/61a86245dcd7acc66fac29348839d5f4dbd219d5)
We will also need a detach button.

acozma, 2018-01-17:
Can you please ask the designer to provide a mockup for a detach button?

DanielGuramulta, 2018-03-19:
The user can also double click a tool in the menu to detach it now, so I will mark this issue as fixed

amiclaus, 2018-03-22:
On Scopy build: 1beb185 (v0.11) the features are implemented.

closed 2018-03-22 by amiclaus