analogdevicesinc/scopy #107: Item 4 from issue#8: screen resizing

ABanut, 2017-05-24:
Item 3 from issue #8 

"This is a recurring issue across all tools. Screen resizing is messed up. In this tool turning on and off the history plot causes the screen to go off the right side of the screen. Can't manually resize window to get it back. Left-Right resize does not work but up-down resize does?"

damercer, 2017-05-24:
The resizing problem with the voltmeter screen was fixed a while ago but now when the window is sized to the minimum allowed left-right width, the right most digit of the display is hidden under the VDC when the history graph is being displayed. When you stretch the window wider the last digit slides from under the VDC. The VDC could stand to be in a smaller font.

Doug

pcercuei, 2017-05-24:
@damercer, only on Windows though. On Linux it looks good at minimum resolution. I need to figure out why this happens; chances are that it doesn't use the same font.

rgetz, 2017-05-24:
if we have cross platform issues - it might be nice for people to see (paste a screen shot into the tracker) platforms they don't use all the time.

commit e72c3f7e (2017-06-08): DMM: Use fixed size font and drop hardcoded color for label

Fixes #107.
closed 2017-06-08 by pcercuei