analogdevicesinc/scopy #108: Item 4 from issue #8: True RMS reading

ABanut, 2017-05-24:
Item 4 from issue #8 

"The AC voltage does not seem to be True RMS as it does not include the DC component. You should consider adding True RMS readings."

damercer, 2017-05-24:
I think there is more than just that wrong with the volt meter. Just now I did the following tests using Scopy a5ce500.
I connected 1+ to W1 and 1- to ground. Set the W1 as DC = 1.5V. Hit run on signal generator.
Confirmed DC 1.5 V on channel 1 with scope tool. 
With voltmeter tool hit run and the display slowly rises ( a few seconds ) and eventually stops at 1.3 V. 
With the generator set to sine wave with 2.828 V amplitude ( 1 V AC RMS ) and 1.5 V DC offset. Confirmed with scope 1.5 V offset 1.75 TRMS and 1.0 V AC RMS.
Voltmeter reads 1.235 DC and 0.839 VRMS.

Things drift around a lot due to huge TC of DAC fullscale but None of the voltmeter readings are correct.

Any way the original request for TRMS readout vs ACRMS is valid.

Doug

pcercuei, 2017-05-25:
I tried it, but it works fine here. I read back the constant 1.5V DC correctly. If I set the 2.828V + 1.5VDC waveform, I read back around 1V AC RMS.

I think it's related to the "hardware offset" feature that was introduced recently. If I move the curves in the oscilloscope, the values I read back in the oscilloscope are completely wrong.

damercer, 2017-05-25:
That may well be true. I had adjusted the scope channel 1 position to center the waveform on the grid.

I think in general the developers need to thinks across the tools when making changes and what effect they may be causing.

pcercuei, 2017-06-05:
I think it does not happen anymore in the latest master, could you confirm?

pcercuei, 2017-06-14:
What frequency do you use for the sine wave?
The DC average of the signal is already done over "some time", but the time is most likely too short.

damercer, 2017-06-14:
That's the point the AC frequency part of the user's signal could be 
almost anything. I was using 1 KHz I think, but 60 / 50 Hz of the AC 
power mains comes to mind say if they are looking at a step down 
transformer based power supply design. Maybe in the DC meter mode the 
sample period needs to be adjustable. It is a trade off between the 
frequency response and the refresh rate. Most DMMs update maybe once or 
twice a second so the average can remove even very low frequencies.

Doug

pcercuei, 2017-06-14:
What would be the upper limit for the frequency, above which the RMS cannot be calculated?
This influences the sample rate that I can use and the number of samples used to calculate the DC offset.

damercer, 2017-06-14:
The frequency is only limited by the hardware. i.e. the bandwidth of M2K.

The real issue is the lower frequency limit. How close to 0 Hz do you want
to support. The lower the frequency the longer the average. You can of 
course
use a moving or sliding average and calculate new values more often.

rgetz, 2017-06-14:
OK - so looking at others:

http://en-us.fluke.com/products/digital-multimeters/fluke-3000-fc-wireless-multimeter.html#techspecs
AC Voltage : 45 Hz to 500 Hz	: 1.0% / 500 Hz to 1 kHz	2.0%

http://literature.cdn.keysight.com/litweb/pdf/5990-3970EN.pdf?id=1701846
AC voltage (true RMS, AC coupling mode) specs are 30 Hz to 50 Hz; 50 Hz to 10 kHz; 10 kHz to 30 kHz; 30 kHz to 100 kHz;

Which I think makes sense - it goes through a 30 Hz, high pass filter. (which removes DC), so you can accurately measure 60Hz.

This might be nice to make this selectable (30 Hz, 60Hz, 120Hz) so in some labs - you can remove power supply or fluorescent lighting hum.

-Robin

-Robin

pcercuei, 2017-07-18:
I implemented two modes, 20 Hz -> 800 Hz and 800 Hz -> 400 kHz. No idea about the accuracy.

Right now, because of #170 the peak-to-peak amplitude set in the signal generator might result completely different, so be sure to check the amplitude in the oscilloscope before testing the voltmeter.

amiclaus, 2017-07-25:
On Scopy build: f759590 (v0.6) the bug is fixed.

closed 2017-07-25 by amiclaus