analogdevicesinc/scopy #110: Voltmeter: Does not know what gain range the analog signal path is set to

damercer, 2017-05-24:
It seems that the Voltmeter tool is not checking to see what gain range the analog signal path is set to. For example with the V/Div for channel 1 set to 1V in the scope tool ( low gain setting ) the voltmeter is sort of right ( see previous issue )  but when you change the V/Div to 500 mV ( high gain setting ) the voltmeter reads a value about 10X higher than it should.

Doug

pcercuei, 2017-05-25:
I think @dNechita is working on something that will (indirectly) fix that.

dNechita, 2017-05-26:
This commit https://github.com/analogdevicesinc/scopy/commit/7780096486a5f0d95d918b639c2a9fe59a365f6b should have fixed the problem. The Voltmeter always uses the highest sample rate and the low gain (wide input range) and triggers set to off.

damercer, 2017-05-26:
> This commit 7780096

For which version of master did this start taking effect? I might be 
using an older version.

rgetz, 2017-05-26:
3 days ago.

amiclaus, 2017-07-24:
On Scopy build: 3aae9de (v0.6) the bug is fixed.

closed 2017-07-24 by amiclaus