analogdevicesinc/scopy #111: Adding 't1' Math Channel generates Runtime Error

amiclaus, 2017-05-25:
Scopy version: [54d0492](https://github.com/analogdevicesinc/scopy/commit/54d049240ff5a83976145d8a7d5058b8c5b0e95c), OS: Windows, Firmware: v0.13

Run Oscilloscope, then add 't1' Math Channel. Runtime Error appears and Scopy crashes.

@pcercuei mentioned that most probably is a [gr-iio](https://github.com/analogdevicesinc/gr-iio) problem.

AlexandraTrifan, 2017-10-31:
This issue was fixed with a commit on the gr-iio repository. The explanations and changes can be found here: https://github.com/analogdevicesinc/gr-iio/pull/14. 
For Windows: the latest Scopy appveyor build should already use the latest gr-iio version.
For Linux: installing the latest version of gr-iio on your Linux machine should fix this.

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the bug is fixed.

closed 2017-11-21 by amiclaus