analogdevicesinc/scopy #112: Number of Math Channels added is not limited properly

amiclaus, 2017-05-25:
Scopy version: [54d0492](https://github.com/analogdevicesinc/scopy/commit/54d049240ff5a83976145d8a7d5058b8c5b0e95c), OS: Windows, Firmware: v0.13

Select Oscilloscope, then start adding math channels. After adding a high number of channels, the interface will look more or less like this: 
[image]

The user won't be able to select and use the channels.

Either the number of channels that can be added is limited or a button is introduced to close all math channels.

pcercuei, 2017-05-25:
I think it makes sense to limit the number of math channels. There shouldn't be a need for more than a few of them.

DanielGuramulta, 2017-05-25:
I`ve made a commit that solves the issue: https://github.com/analogdevicesinc/scopy/commit/b7e680566c3e0b09a249e289593e5f78f5d5b7f1.

pcercuei, 2017-05-25:
I think you could just close the math panel and disable the '+' button (hide it) when you reach the maximum number of math channels, then it's pretty obvious to the user that the limit has been reached.

pcercuei, 2017-05-25:
Good work, but it disables the "Add channel" button, could you make it hide the ui->btnAddMath button instead? And close the math panel menu if it's currently opened.

DanielGuramulta, 2017-05-26:
Now it will close the math panel: https://github.com/analogdevicesinc/scopy/commit/39f4b7bce1e17ee77e6d80ccf40d1b4f80d5fac2

pcercuei, 2017-05-26:
But it opens the channel settings 😁

commit f71a8eec (2017-05-26): Oscilloscope: Add a limit for the math channels

When the limit is exceeded we will hide the addMathChannel
button and the math panel on the right. When we will delete a channel
the add button will be visible again.
Fixes #112
commit 6915e709 (2017-05-26): Oscilloscope: Add a limit for the math channels

When the limit is exceeded we will hide the addMathChannel
button and the math panel on the right. When we will delete a channel
the add button will be visible again.
Fixes #112
closed 2017-05-26 by pcercuei