analogdevicesinc/scopy #113: Change Time Base through +/- buttons crashes Scopy

amiclaus, 2017-05-26:
Scopy version: [6915e70](https://github.com/analogdevicesinc/scopy/commit/6915e709e99630b7cf8d243b099c23b0772d999f) , OS: Windows, Firmware: v0.13

Delete Scopy.ini file first, then open Scopy, select Oscilloscope -> Channel Settings. Press + or - button corresponding to Time Base. The application crashes.
[image]

[image]

dNechita, 2017-05-28:
Fixed in commit https://github.com/analogdevicesinc/scopy/commit/e653e6f0f987405e1b97f9eac602e7a77ee0af47 and in https://github.com/analogdevicesinc/scopy/commit/c03357032446589543726c16b5f3e6260598e8e8 (by default, with no .ini to load the time position should be 0 not -5 seconds).

amiclaus, 2017-05-29:
Fixed via [e653e6f](https://github.com/analogdevicesinc/scopy/commit/e653e6f0f987405e1b97f9eac602e7a77ee0af47) & [c033570](https://github.com/analogdevicesinc/scopy/commit/c03357032446589543726c16b5f3e6260598e8e8)

closed 2017-05-29 by amiclaus