analogdevicesinc/scopy #115: PG: Scopy Minimum Size window make texts and buttons overlap from Issue #60

amiclaus, 2017-05-26:
Select Pattern Generator Feature and scroll inside the plot area until overlap is obtained as shown in the picture below:
[image]

AlexandraTrifan, 2017-05-26:
This happens in the Logic Analyzer as well. The issue should be fixed after commit 966468798a0ee13d035d605da43935b597e6fdf3. (The initial fix was introduced with commit ce616a6d2a7987f5ee24f726c57a621ae3c36815.)

AlexandraTrifan, 2017-05-26:
The previous commits did not solve the problem completely. Commit 93f7e5e7c977e049b8c9e0c2198e9f431ce80bd0 introduces another fix.

amiclaus, 2017-05-29:
Fixed via [93f7e5e](https://github.com/analogdevicesinc/scopy/commit/93f7e5e7c977e049b8c9e0c2198e9f431ce80bd0)

closed 2017-05-29 by amiclaus