analogdevicesinc/scopy #117: scopy-setup.exe protection issue

damercer, 2017-05-26:
The copy of scopy-setup.exe I downloaded on 5/15/17 has some sort of super protection set such that I can't delete it / or rename it. I would like to download the latest version but I would like to get rid of ( or at least rename ) this copy from my computer system first. This is on the ADI IT controlled laptop. When looking at the properties it says the owner is unknown and it will not let me set it to myself.

So How do I get rid of this thing!

Doug

dNechita, 2017-05-28:
There might be some Windows restrictions that prevent you from modifying installed programs located at some Windows predefined paths like 'Program Files', etc.
You should be able to uninstall the application from Start->All Programs->Scopy->Uninstall Scopy  or from Windows 'Programs and Feature'.

Scopy can be installed [locally](https://github.com/analogdevicesinc/scopy/blob/master/README.md) and you could do it in a location where you have full control.

-Dan

pcercuei, 2017-05-29:
I blame ADI's security. I've been having the same problem with .exe files I compiled myself, moved to C: and then couldn't delete anymore.

damercer, 2017-05-29:
I went back a while later and the file had disappeared on it's own. Must 
have been in some sort of limbo state. The browser must have started to 
delete it as part of the new download timed out and failed. Eventually 
the OS actually got around to deleting the file. Strange..... Anyway I 
was able to download a new version.

closed 2017-05-30 by damercer