analogdevicesinc/scopy #119: Pattern Generator Main Window Improvements

amiclaus, 2017-05-29:
Scopy version: [e56a9d9](https://github.com/analogdevicesinc/scopy/commit/e56a9d9ab3d8c26f97c7d1c584e1e61471d6f292), OS: Windows, Firmware: v0.13

Open the Pattern Generator, and press Run button. The button color changes suggesting that the tool is running (but the Text Label does not change to "Stop" as for other features). 
[image]

In this case nothing happens since no channel was selected/enabled and there is no information provided to the user to enable channels of the pattern generator. (maybe a pop-up/textbox should appear: "Please enable at least one channel").

On the second hand, I think a more suggestive Text Label (Ex. "Enable") should be applied corresponding to the Enable/Disable Channel column rather than "View", since the column is more than a visibility option.

[image]

pcercuei, 2017-05-29:
Please don't work on the UI right now - I have a lot of changes in my global-stylesheet branch.

adisuciu, 2017-06-20:
https://github.com/analogdevicesinc/scopy/commit/4ad1da00cbdeee211d173938a180582c2fd400e6 - fixes label in channel manager
https://github.com/analogdevicesinc/scopy/commit/49ffe7041205d70c736cf7ac470975dfe0dac125 - adds a status label and changes how patterns are generated. It is no longer needed to start-stop-start the patterngenerator when modifying a pattern.

amiclaus, 2017-07-06:
On Scopy build: 21bb304 the bug is fixed.

closed 2017-07-06 by amiclaus