analogdevicesinc/scopy #122: Spectrum Analyzer: Frequency sweep settings

damercer, 2017-05-31:
You have start, stop, center and span controls. All four seem to be operational at the same time. This makes it very hard to change the settings. Generally in a spectrum analyzer only Start / Stop OR Center / Span frequency controls are used. These control settings are either / or not both at the same time. Trying to get all four to agree at the same time as you change one of them is next to impossible.

This might be a separate issue but I assume you will eventually change the "resolution bandwidth" automatically based on the minimum frequency or allow the user to set it?

I also assume you will eventually have controls for adjusting the vertical scale ( dB/Div ).

And as I've already pointed out in the Network analyzer the frequency scale needs to be Log as default ( optional linear scale ).

I'm sure there are many other things not yet implemented as well.

Doug

adisuciu, 2018-01-23:
In the latest version of Scopy, you either set Start/Stop or Center/Span. This way you can set the range of the instrument. 
Resolution bandwidth can be selected by the user.
Controls for selecting vertical scale are implemented.

Also in the network analyzer, the frequency scale is log by default, with linear scale optional.

amiclaus, 2018-03-12:
On Scopy build: 6bd7075 (v0.11) the bugs are fixed.

closed 2018-03-12 by amiclaus