analogdevicesinc/scopy #123: Scopy release v0.4 crashes on Windows 7

damercer, 2017-05-31:
Hi:
This is very strange. Why would Scopy crash on this ADI Windows 7 Dell lap-top all of a sudden and not on the Windows 7 desk-top? I've tried re-installing both the released v0.4 and a newer build from master several times in different places on the C: dive, under Program Files ( default ) and at the top of C:. Scopy runs but when I click on the connect button after selecting the board that is plugged in, it takes a few seconds and crashes. The same combination of Scopy and m2k works fine on the windows 7 Dell desk-top????

An older version of Scopy had been working fine up until I replaced it with v0.4.

Doug

mhennerich, 2017-05-31:
Hi Doug,

Can you try to delete C:\Users\YOU\AppData\Roaming\ADI\Scopy.ini

I had a similar problem recently and that fixed it.

-Michael

damercer, 2017-05-31:
Yup, That fixed it. Thanks much.
If this sort of thing can happen often maybe it should be written up in a FAQ in the Scopy page (user guide) on the ADI Wiki? I'm guessing there will be other things like this that come up over time.

Doug

mhennerich, 2017-06-01:
Yes this should go into the FAQ. And we also need to make sure that installing a new Scopy version will delete this file too. Because I think INI incompatibility between versions caused this issue.

-Michael

damercer, 2017-06-01:
My thought exactly. If Scopy has trouble reading the init file ( or any 
other for that mater ) it should gracefully skip over the problem and 
move on.

Doug

rgetz, 2017-06-01:
Doug:
Yes - it's a bug, I changed the title, and whoever updates the installer can close this.

dNechita, 2017-06-07:
@adisuciu @AlexandraTrifan @pcercuei and myself: To fix and close this issue we need to identify which property in the .ini file causes the crash and fix it. Could be from any instrument.

As mentioned on previous comments, we need to make sure that loading a file doesn't crash the application regardless of the content of the file.

@rgetz I'm not sure what you meant by updating the installer. If we make sure Scopy gracefully skips over problems then the installer doesn't need to delete the old .ini file, which is better since users get to keep their settings between two installs.

rgetz, 2017-06-07:
@dNechita : For now (while in development) I don't think users would be that upset if between installs, they lost their configurations. After 1.0 (when ever that happens) - I totally agree, but things are too much in flux for now to spend time tracking things down like this (I think).

Thoughts?

pcercuei, 2017-06-07:
Crashes occur when restoring from the INI - but the crash is probably triggered by a single property, so we can just catch the exception and continue restoring the other properties like nothing happened. Next time Scopy quits, the property will be properly overwritten with a proper value that doesn't make everything crash.

amiclaus, 2018-02-05:
Since we reached v0.10 and the bug is fixed on this release version, I will close the issue.

closed 2018-02-05 by amiclaus