analogdevicesinc/scopy #125: Scope: Order things are drawn on the screen

damercer, 2017-06-02:
In the waveform display area, the blue horizontal position line needs to be drawn behind the waveform trace. As we can see in this screen shot the rising edge of the trace is obscured by the blue line.

[image]

This would also be true for the other horizontal position lines and maybe the cursor lines as well. As a general case the waveform traces should be drawn on top of everything else ( i.e. last I'm guessing ).

Doug

DanielGuramulta, 2018-01-17:
Fixed with commit : https://github.com/analogdevicesinc/scopy/commit/10de3a9934c74771870fd9abf1b030f9d1058f39

amiclaus, 2018-01-18:
On Scopy build: d85de1c (v0.10) the feature is implemented.

closed 2018-01-18 by amiclaus