analogdevicesinc/scopy #127: Osc: Adding math channels over the limit

DanielGuramulta, 2017-06-15:
1) When the limit is reached the math panel would close but clicking on the btnSettings would open it again:
[image]
2) When the limit is reached, but we click multiple times on the add math channel button in the math panel the limit will be exceeded:
[image]

DanielGuramulta, 2017-06-15:
This commit should solve the first issue: https://github.com/analogdevicesinc/scopy/commit/c9fdf1b5ff5cc8cd83fa5d2c26136f3ff0e58020
This one should solve the second: https://github.com/analogdevicesinc/scopy/commit/1c12c805cbdc5ddacb6a1859212bb088c1c2f8b2

pcercuei, 2017-06-15:
The first commit doesn't fix the problem. The panel doesn't open up, but the channel settings option button is shown as pressed.

pcercuei, 2017-06-15:
You can fix the two with a simpler change: https://gist.github.com/pcercuei/375334314e11a1756acdcfd2fa2de693

commit 202ffcbf (2017-06-19): OSC: The math panel will not be opened again after the limit is reached

Now clicking on the settings button, after the limit for the math
channel is reached, will not open the math panel again.
Fixes #127
commit 0501fdae (2017-06-20): OSC: The math panel will not be opened again after the limit is reached

Now clicking on the settings button, after the limit for the math
channel is reached, will not open the math panel again.
Fixes #127
closed 2017-06-20 by dNechita