analogdevicesinc/scopy #130: Scopy - System Error

amiclaus, 2017-06-21:
When I try to run Scopy build either from an own build or from an appveyor build, the following error pops-up:
[image]
In my C:\msys64\mingw64\bin there is only "libpython3.6m.dll".
I've tried to update the environment variables to python3.6 but that doesn't solve the problem.

DanielGuramulta, 2017-06-21:
Renaming the dll to libpython3.5m will let Scopy run.

dNechita, 2017-06-21:
The reason this is happening is that MSYS2 switched from Python 3.5 to 3.6. 

We need to update some of the files from the bundle that we have stored at http://swdownloads.analog.com/cse/build/msys64.tar.xz,.
I think that libsigrokdecode-3.dll needs to be recompiled by taking into account the new version of Python.

pcercuei, 2017-07-04:
So this bug itself is fixed, but it doesn't work 100% right yet, see #143.

amiclaus, 2017-07-06:
On Scopy build: 21bb304 the bug is fixed.

closed 2017-07-06 by amiclaus