analogdevicesinc/scopy #133: Group Channels not working properly

amiclaus, 2017-06-28:
Scopy Version:  fb92e653; OS: Windows; Firmware v0.13;

Make a loopback between Channel Group DIO0-7 and DIO8-15
[image]

Set Group DIO0-7 as output with value 0 and DIO8-15 as input. Run the instrument. 
[image]
Change slightly the value of DIO0-7:
[image]

There is no proper correlation between the 2 channel groups.

adisuciu, 2017-06-28:
@larsclausen  @acostina This seems to be an HDL/firmware issue. The issue can be reproduced by doing the following in iio-scope

```
setup - short channel 0 to channel 8
device - m2k-logic-analyzer

set channel 0 direction output
set channel 0 raw 1
read channel 8 raw 1
set channel 0 direction input

set channel 0 direction output
set channel 0 raw 0
read channel 8 - raw - 0

set channel 0 direction input
set channel 0 raw 0
read channel 8 - raw - 1  - mismatch
```

lclausen-adi, 2017-06-28:
Hm, so the problem is when both channel 0 and channel 8 are set to input they don't read the same value for raw?

adisuciu, 2017-06-28:
Sorry, typo, updated test steps

mhennerich, 2017-06-28:
Works without issues for me.
Tried it with the latest firmware image I just shared.

-Michael

adisuciu, 2017-06-28:
I flashed latest firmware, still get the same behavior. I should mention I still use rev A. board. I created small test program to reproduce the error. 

`dev = iio_context_find_device(ctx, "m2k-logic-analyzer");
	auto ch0 = iio_device_find_channel(dev,"voltage0", 0);
	auto ch8 = iio_device_find_channel(dev, "voltage8", 0);
	char buffer[10];

	iio_channel_attr_write(ch0, "direction", "in");
	iio_channel_attr_write(ch8, "direction", "in");

	printf("set channel0 direction output\n");
	iio_channel_attr_write(ch0, "direction", "out");
	printf("set channel0 raw 1\n");
	iio_channel_attr_write(ch0, "raw", "1");
	iio_channel_attr_read(ch8, "raw", buffer, 10);
	printf("read channel 8 - %s\n",buffer);
	printf("set channel 0 direction input\n");
	iio_channel_attr_write(ch0, "direction", "in");

	printf("set channel0 direction output\n");
	iio_channel_attr_write(ch0, "direction", "out");
	printf("set channel0 raw 0\n");
	iio_channel_attr_write(ch0, "raw", "0");
	iio_channel_attr_read(ch8, "raw", buffer, 10);
	printf("read channel 8 - %s\n", buffer);
	printf("set channel 0 direction input\n");
	iio_channel_attr_write(ch0, "direction", "in");

	printf("set channel0 direction output\n");
	iio_channel_attr_write(ch0, "direction", "out");
	printf("set channel0 raw 0\n");
	iio_channel_attr_write(ch0, "raw", "0");
	iio_channel_attr_read(ch8, "raw", buffer, 10);
	printf("read channel 8 - %s\n", buffer);
	printf("set channel 0 direction input\n");
	iio_channel_attr_write(ch0, "direction", "in");`

mhennerich, 2017-06-28:
I also have RevA. And I guess (hope) you're NOT shorting the outputs like shown in the picture above!
So only DIO0<->DIO8, etc. ...

-Michael

adisuciu, 2017-06-28:
I shorted just the digital inputs. I also tried shorting only 0 to 8, same behavior.

lclausen-adi, 2017-06-29:
It's and issue with the HDL and will be fixed. For now as a workaround you need to set raw to 1 temporarily to have the direction change applied.

E.g.

direction = out
raw = 1 // Direction change happens
raw = X

lclausen-adi, 2017-06-29:
Fixed by https://github.com/analogdevicesinc/hdl/commit/8755e6da44dca11dfab18765485f77bd516f9478

closed 2017-06-29 by lclausen-adi