analogdevicesinc/scopy #135: Power Supply - Number of decimal places in set values

tagoylo, 2017-06-29:
Scopy build : [1839ee5d](https://github.com/analogdevicesinc/scopy/commit/1839ee5d7a90b30ff9221116ac4fd61f20edc389);  OS: Windows ; Firmware: v0.13-dirty

In the positive/negative output control, the user can set values up to 5 decimal places but the displayed LCD set value is only up to 3 decimal places. I think it would be less confusing for the user to have it in uniform; that the value they set will also reflect to the value shown on the display. 

[image]

pcercuei, 2017-07-10:
@dNechita, I think we should limit the number of decimals in the custom spinbox object to 3.

dNechita, 2017-07-10:
With this commit https://github.com/analogdevicesinc/scopy/commit/1b52ce4ea52bc5e9b5b8b73a3e4ea9c318d1a1a1 all spinboxes have a maximum of 3 decimals. The decimal count can be changed individually if some instruments need to display more or less decimals.

amiclaus, 2017-07-24:
On Scopy build: 3aae9de (v0.6) the bug is fixed.

closed 2017-07-24 by amiclaus