analogdevicesinc/scopy #138: Logic analyzer: Setting a trigger will shift the view to show the lower pins if they're not shown on the screen

drbautista, 2017-06-30:
OS: Windows 7 64bit
Scopy Version: 1839ee5
Firmware Version: v0.13

Open Logic Analyzer instrument

1. Set any DIO’s trigger while DIO15 are not shown on the screen

[image]

2. The view will shift to show the lower end pins. It's not very convenient to keep on scrolling up when you still want to set the trigger of all the upper pins.

"This kind of  feature should be present when “group with selected” button is clicked since the group is initially created below."

AlexandraTrifan, 2017-06-30:
Fixed with commit 9fd35a5e2f35f81252b1539eef712813b1c8e249.
Whenever a trigger is set, the position of the scroll area stays the same. 
In addition, when a group is created with the 'Group with selected' button, the scroll area changes its position to display the newly added group, at the bottom of the list.

amiclaus, 2017-07-05:
On Scopy build: 1e648f6 the bug is fixed.

closed 2017-07-05 by amiclaus