analogdevicesinc/scopy #140: Overlapping values in graph when in minimum window size

tagoylo, 2017-06-30:
Scopy build :[1839ee5d ](https://github.com/analogdevicesinc/scopy/commit/1839ee5d7a90b30ff9221116ac4fd61f20edc389) OS: Windows ; Firmware: v0.13-dirty

Values in the horizontal axis of the graph overlaps when settings button is clicked and the panel on the right opens. This happens when window is in minimum size. This does not appear when in full screen. The issue is illustrated in the picture below.

[image]

DanielGuramulta, 2017-09-06:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/86a7d7ee270d49152efddc29f8f0a336202b01f5

amiclaus, 2017-09-08:
On Scopy build: 152efe0(v0.7) the bug is fixed.

closed 2017-09-08 by amiclaus