analogdevicesinc/scopy #142: Color Settings doesn't change accordingly with selected channel

tagoylo, 2017-06-30:
Scopy build : [1839ee5d](https://github.com/analogdevicesinc/scopy/commit/1839ee5d7a90b30ff9221116ac4fd61f20edc389) OS: Windows ; Firmware: v0.13-dirty

1. Change the color settings of one of the channels, in the illustration I've change DIO0's color settings.
2. Select another channel, the color settings will only show the colors set for DIO0. It will only change when you move your cursor to the colors or when you click again somewhere near the color settings or panel.

[image]

AlexandraTrifan, 2017-07-03:
Fixed with commit d31dd85250f592045e0066b0436fff5d1d633492.

amiclaus, 2017-07-05:
On Scopy build: 1e648f6 the bug is fixed.

closed 2017-07-05 by amiclaus