analogdevicesinc/scopy #144: Voltmeter: When a channel is in DC mode, switching the other channel's mode will reset the measurement

drbautista, 2017-07-04:
OS: Windows 7 64bit
 Scopy Version:  3021dec
 Firmware Version: v0.13

Open Voltmeter instrument:

1. Set Channel 1 to DC mode
2. Connect scope ch+ to positive power supply and scope ch- to gnd.
3. Set the positive power supply to any voltage level, e.g. 3V
4. Switch channel 2's mode from DC to AC or AC to DC.

Result: Channel 1's voltage reading will reset to 0 and settle again to the voltage input. Channel 2 has a similar behavior in DC Mode.

This behavior doesn't happen in AC Mode.

[image]
[image]

pcercuei, 2017-07-04:
Long story short: we have a hack in place (https://github.com/analogdevicesinc/scopy/blob/master/src/iio_manager.hpp#L90) that used to be a workaround for a bug in GNU Radio. The bug was fixed, but it appears that the hack is still needed, otherwise we get lockups in the data stream. This is a side effect of that hack.

amiclaus, 2018-02-05:
Update: the bug is still valid on Scopy v0.10.

adisuciu, 2020-10-19:
This was fixed at some point.

closed 2020-10-19 by adisuciu