analogdevicesinc/scopy #148: Single Button available while Instrument is running

amiclaus, 2017-07-06:
Scopy Version: a90e179
For the Oscilloscope and Logic Analyzer when the Instrument is running, the single shot button is grey and unavailable. 
For the Pattern Generator the user can take a single shot while the instrument is running. After the single shot is made, the instrument stops.

There should be the same approach for all 3 instruments.

dNechita, 2017-08-30:
The following changes provide the functionality that was requested.
Oscilloscope:
https://github.com/analogdevicesinc/scopy/commit/d75cd14cd9b9e5f720d1ae409dc92c8cf8851cf0
Spectrum Analyzer:
https://github.com/analogdevicesinc/scopy/commit/1abd0f865d5047a9633f89e45bd7ea87df961c84

AlexandraTrifan, 2017-08-30:
This was implemented in the Logic Analyzer with commit 2d652d64565ae3f3351cef40e1139a6c6bb2b706.

amiclaus, 2017-08-30:
On Scopy build: 5bb6466 (v0.7) the enhancement is implemented.

closed 2017-08-30 by amiclaus