analogdevicesinc/scopy #151: Missing drivers after Scopy installation on Windows

msuciu, 2017-07-06:
After Scopy installation some drivers are not installed, CDC Serial, IIO and RNDIS. This drivers are installed when PlutoSDR-M2k-USB-Drivers.exe is executed https://github.com/analogdevicesinc/plutosdr-m2k-drivers-win/releases, but there is no information on Scopy  regarded on this missing drivers. I thing it will be a good idea to include this drivers on installer, or add some information on Scopy home tab/page, or a popup message about this to redirect/inform the user to install this drivers too.
[image]

pcercuei, 2017-07-06:
Scopy is not supposed to be M2K-specific (even if it kind of is right now), but a generic tool.
Installation of Scopy and the drivers should be separate.

msuciu, 2017-07-06:
Ok, I understand, but the user should be informed somehow about the missing drivers. Another option can be to add this information on Wiki.

amiclaus, 2018-05-09:
Information about M2K missing drivers was added recently on the Scopy main wiki page:
https://wiki.analog.com/university/tools/m2k/scopy

msuciu, 2018-05-09:
Fix confirmed.

amiclaus, 2018-05-09:
Issue closed.

closed 2018-05-09 by amiclaus