analogdevicesinc/scopy #154: Decoder choices can't be fully read in dropdown when in group

tagoylo, 2017-07-07:
Scopy build: [a90e179](https://github.com/analogdevicesinc/scopy/commit/a90e1793fd2ab7f9d490b170afc27a1ae429e671)
OS: Windows
Firmware: v0.13-dirty

When you click on the drop-down for decoder when in group, the list of choices cannot be seen/read fully, especially to the longer terms. Maybe add a  horizontal scroll bar  in order for the choices to be read fully. This happens even in full screen.

[image]

AlexandraTrifan, 2017-07-10:
Fixed with commit f87f3173979411f2aeefc3d6ef9d42a7d556b414.

amiclaus, 2017-07-24:
On Scopy build: 3aae9de (v0.6) the bug is fixed.

closed 2017-07-24 by amiclaus