analogdevicesinc/scopy #155: Reset Instrument button doesn’t reset trigger selection to None

tagoylo, 2017-07-07:
Scopy build: [a90e179](https://github.com/analogdevicesinc/scopy/commit/a90e1793fd2ab7f9d490b170afc27a1ae429e671)
OS: Windows
Firmware: v0.13-dirty

Set trigger to any of the selection in the dropdown, change also the name of the channel, disable  other channels, then after setting, click Reset Instrument. The name will change to default and also the disabled channels will be enabled again, as default. But the trigger selection does not change back to None, it stays as is.

[image]

AlexandraTrifan, 2017-07-07:
Fixed with commit 2ce436f18e479c44fdcca2bccb8a498e59dc9de2.

amiclaus, 2017-07-24:
On Scopy build: 3aae9de (v0.6) the bug is fixed.

closed 2017-07-24 by amiclaus