analogdevicesinc/scopy #156: "Show All" doesn’t change back to "Hide Inactive" after Reset Instrument

tagoylo, 2017-07-07:
Scopy build: [a90e179](https://github.com/analogdevicesinc/scopy/commit/a90e1793fd2ab7f9d490b170afc27a1ae429e671)
OS: Windows
Firmware: v0.13-dirty

Disable other channels.
Click “Hide inactive”.
Then click “Reset Instrument”.

The hidden channels will show up again as you click reset instrument. But the Show All button will stay as is even if the hidden channels are not anymore hidden. It doesn’t change back to Hide Inactive. 

[image]

AlexandraTrifan, 2017-07-07:
Fixed with commit 309d4ba949a3449f0df845f14117efc27d490f5a.

amiclaus, 2017-07-24:
On Scopy build: 3aae9de (v0.6) the bug is fixed.

closed 2017-07-24 by amiclaus