analogdevicesinc/scopy #157: Channel settings position value

msuciu, 2017-07-07:
When trigger position is in right side on the screen the position value should always be a negative value, but if you move the trigger with the mouse cursor that value is sometimes positive.
[image]
[image]

dNechita, 2017-07-10:
In reality, the value doesn't jump from positive to negative (it only appears to do that). Because there are too many decimals and not that much space, the entire number gets shifted to the left and hides the minus sign.
This commit fixes the issue: https://github.com/analogdevicesinc/scopy/commit/1b52ce4ea52bc5e9b5b8b73a3e4ea9c318d1a1a1

amiclaus, 2017-07-24:
On Scopy build: 3aae9de (v0.6) the bug is fixed.

closed 2017-07-24 by amiclaus