analogdevicesinc/scopy #159: Settings button located at the upper right corner negates use of the select button

tagoylo, 2017-07-07:
Scopy build: [a90e179](https://github.com/analogdevicesinc/scopy/commit/a90e1793fd2ab7f9d490b170afc27a1ae429e671)
OS: Windows
Firmware: v0.13-dirty

Select cursor settings and open it
Deselect cursor settings. This will close the panel
Click on the settings button located at the upper right corner of the interface. The cursor settings panel will open even if it has been deselected. 

This also happens with the measure settings. It negates the use of the select button. The behavior of the settings button on the upper right corner is that it remembers the last settings used and acts like it.

[image]

DanielGuramulta, 2017-08-21:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/cc1870280c20eb9dd85bfdd0952831f1c1a99851

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus