analogdevicesinc/scopy #160: Display cursor arrea

msuciu, 2017-07-07:
When cursors are enabled I guess it will be a good idea to make the background transparent to see the signal behind, or to have the option to move the cursor window somewhere else in plot.

[image]

damercer, 2017-07-07:
I agree the area around the cursor text should be "transparent" so as not to block the waveforms. 
There could be four options to place the text in any of the four corners of the display, Upper left/right
lower left/right.

Doug

DanielGuramulta, 2017-09-22:
I've implemented the above enhancement with commit: https://github.com/analogdevicesinc/scopy/commit/2c2c7ff5022a58814aa435887ecae923bbd5a1c3

amiclaus, 2017-10-10:
On Scopy build: 8f7e63f (v0.8) the the features are implemented.

closed 2017-10-10 by amiclaus