analogdevicesinc/scopy #161: Reset Vertical and Horizontal position by double click

msuciu, 2017-07-07:
When you double click on the trigger or channel position, the position should be reset at 0.

[image]

damercer, 2017-07-07:
I like this idea. It would be convenient to have a quick way to reset things to a know state.

Doug

AlexandraTrifan, 2017-11-06:
This was added with commit cf98b0f71cbe2d53bcaa40c3d76e0cb6a81854c5 .

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the enchancement is implemented.

msuciu, 2018-05-21:
[image]

amiclaus, 2018-05-21:
It's really strange. This issue is valid only sometimes when Scopy horizontal window size is different than minimum or maximum. It happens both for Oscilloscope and Logic Analyzer

amiclaus, 2018-07-31:
Update: The issue persists for custom Scopy window size.

DanielGuramulta, 2018-10-08:
Should be fixed with commit: https://github.com/analogdevicesinc/scopy/commit/1a7674ab612d1733ce9d782087f5f1bd8dd50c67

amiclaus, 2018-10-11:
On Scopy build (v1.0.2): d6b1429 the bug is fixed.

closed 2018-10-11 by amiclaus