analogdevicesinc/scopy #162: New tracking option

msuciu, 2017-07-07:
When "Tracking" option is enable on the Power Supply you only have the possibility to correct the negative supply with a percentage value. Using this option you don't cover all possibilities, for example is not possible to source +4 Volts and -4.5 Volts. This possible issue can be solved by changing the range of the percentage slider to be between 1-200% if positive voltage is not higher than 2.5 Volts, for example. 

[image]

damercer, 2017-07-07:
I agree, the simplest solution is to allow percentages greater than 100%.

Doug

pcercuei, 2017-07-17:
I don't think that would work, you would want to go above 200%. So what would be a sane maximum value?

I suggest that instead we have a selector for the reference channel, and keep the [0,100]% ratio for the other channel.

rgetz, 2017-07-17:
the sane "max" value - is a percentage that doesn't exceed the output max (+/-5V).

if the output is at 1V, then the max percentage is 500% (output -5V), If the user sets it to 500%, then it means the range on the 1V side is between 0 and 1V, and you can't go higher.

if the output is at 1V, then the max percentage is 500% (output -5V). If the user sets it to 250%, then it means the range on the 1V side is between 0 and 2V, and you can't go higher. If you want to go higher, you must reduce the tracking before.

Does that make sense?

pcercuei, 2017-07-17:
Not really. What if the output is at 0.1V? Then the max percentage is 5000%? That wouldn't work very well.

msuciu, 2017-07-18:
What if you extend the percentage slider range to be between -100 and 100 and add another slider or something to increase the resolution like 1, 0.1, 0.01, 0.001?

mbanciso, 2017-07-18:
I wouldn't complicate the menus too much.
If you need more clicks to use the tracking mode it defeats the purpose.(you are better off entering 2 values in independent mode)

msuciu, 2017-07-18:
Ok, then without adding another slider and just increasing the range between -100 to 100 with a higher resolution should solve the problem.

closed 2017-09-20 by adisuciu