analogdevicesinc/scopy #163: Extend right click option for default zoom

msuciu, 2017-07-07:
If generated waveform is moved on horizontal or on vertical axe or is zoomed in or out is not possible to go on the default zoom only if you zoom in the waveform using left click and than press right click to zoom out on the default zoom. Since is possible to come back on the default zoom using right click when you zoom in using left click, I guess is easy to extend this functionality when you zoom in, zoom out using the mouse scroll or move the waveform using left click.

Also it will be a good idea to limit the horizontal and vertical axe ranges.  

[image]

adisuciu, 2017-10-20:
This is implemented with the latest signal generator preview rework.f2876e55e1d1ece4a1b2f3d358a623d9a0368e55

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the enchancement is implemented.

closed 2017-11-21 by amiclaus