analogdevicesinc/scopy #164: Zoom option using left and right click

msuciu, 2017-07-07:
I guess it will be a good idea to add the option to zoom in using left click for and zoom out using right click, basically an extension of the Sweep menu.

rgetz, 2017-07-07:
I thought we were going to try to stay away from left/right clicks, which don't work on touchscreens, or single button MACs.

I would rather work on pinch/zoom:
https://doc.qt.io/archives/4.6/multitouch-pinchzoom.html

Dan was asking for a 4k touchscreeen for his next demo/tradeshow.

-Robin

dNechita, 2017-07-10:
The plan is to avoid right clicks on the entire application and figure out other ways to do the zooming.
The Spectrum Analyzer is still work in progress and zooming is already planned to be implemented.

closed 2017-07-10 by dNechita