analogdevicesinc/scopy #165: IN channel doesn't respond during run

tagoylo, 2017-07-10:
Scopy build: [a90e179](https://github.com/analogdevicesinc/scopy/commit/a90e1793fd2ab7f9d490b170afc27a1ae429e671)
OS: Windows
Firmware: v0.13-dirty

Steps to reproduce:

Set Ch0 as output, set logic value of 0.
Set Ch7 as input
Click Run

Result:
Ch7 will not respond to what the logic value set is ch0, it will only change when clicking the logic value selector again.
[image]

adisuciu, 2017-07-10:
This sounds like issue #133. It should be fixed with latest firmware version. Can you recheck with version 0.14 ?

amiclaus, 2017-07-25:
On Scopy build: f759590(v0.6) the bug is fixed.

closed 2017-07-25 by amiclaus