analogdevicesinc/scopy #166: Value overlaps unit after turning on history when reading in min window size

tagoylo, 2017-07-10:
Scopy build: [a90e179](https://github.com/analogdevicesinc/scopy/commit/a90e1793fd2ab7f9d490b170afc27a1ae429e671)
OS: Windows
Firmware: v0.13-dirty

Steps to reproduce:

Open scopy power supply,then monitor using voltmeter. Set any value on the power supply.

History plot is on, by default.
Turn off history plot.
Then turn it on again.

The value now overlaps the unit. Unlike before doing the steps.The size of the history plot also is now bigger. This only happens when the voltmeter is reading. And also when window size is at minimum.

[image]

DanielGuramulta, 2017-08-29:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/2a500d89566c81e2c335e13b1bc9c3443059a1b9

amiclaus, 2017-08-30:
On Scopy build: 5bb6466 (v0.7) the bug is fixed.

closed 2017-08-30 by amiclaus