analogdevicesinc/scopy #167: Spikes appear on the Network Analyzer plot

amiclaus, 2017-07-10:
Scopy version: 4b0a59a, OS: Windows 7, Firmware: v0.14

Connect a simple RC HPF (Fc ~ 340Hz) to the Network Analyzer.
Make a sweep between 100Hz-10MHz.

Random spikes appear on the plot (maybe repeat the sweep several times to get the spikes).
[image]

pcercuei, 2017-07-10:
Could you move the "sweep gets stuck" thing in its own bug report?

amiclaus, 2017-07-10:
Sure

pcercuei, 2017-07-10:
The spikes appear when the buffer size of the ADC does not contain an integer number of periods. There is already a mechanism in place for the DAC to ensure that the buffer will contain an integer number of periods, it should be adapted for the ADC path as well.

rgetz, 2017-07-10:
When performing an FFT on data, use a window function - it will make sure that it doesn't matter if you have an exact number of periods.

damercer, 2017-07-10:
To Robin's suggestion of using a window function for the FFT. If you don't go with picking frequencies that fall in the center of an FFT bin you will need to use a flat-top FFT window to get accurate amplitude and phase results. Not all window function are created equal.

rgetz, 2017-07-10:
Yes; I should have been more specific. Math/equation at:

https://en.m.wikipedia.org/wiki/Window_function#Flat_top_window

damercer, 2017-07-10:
There is Python code that calculates a flat-top window ( and others ) in 
ALICE if that is any help.

Doug

pcercuei, 2017-07-11:
So we actually don't use a FFT there, we just cross-correlate with a cosine.

rgetz, 2017-07-11:
Most cross correlations use an FFT, a cross product, and an IFFT. A window function can still be used. 

What's the algorithm for your xcorr?

pcercuei, 2017-07-11:
No idea :)
I just use the complex multiply block from GNU Radio.

rgetz, 2017-07-11:
To do a proper cross correlation, you can't use a simple multiply. 

Try something like:
http://www.olifantasia.com/projects/gnuradio/mdvh/passive_radar/correlator.py

pcercuei, 2017-07-11:
Well, I guess I don't cross-correlate, then.
This is more or less the flowgraph that Scopy generates:

[image]

damercer, 2017-07-11:
Not to get too deep in the weeds on this but it would seem to me that just doing an FFT and getting the amplitude and phase ( difference between input and output ) would be simpler.
You already know what frequency the peak should be at but you can easily search the array to
find it. I'm assuming that in the Spectrum Analyzer you are doing FFTs and you should just reuse that stuff while sweeping the sine source.

Doug

lclausen-adi, 2017-07-11:
The current code essentially does a FFT for a single bin. But it is more precise since we can choose the center frequency of the bin freely to match the frequency of the sweep tone.

rgetz, 2017-07-11:
Hate to say it - but if it was more precise - we wouldn't be seeing the spikes/errors in the data :)

I have seen lots worse than the above when looking at things.

I think it is independent FFT on both signals, (assuming you have 10-100 periods or more of the waveform) find the peaks in both FFTs, and take a ratio of amplitudes, and a simple subtraction for phase difference. That may be more computationally complex, but should eliminate the errors (be less sensitive to full periods on buffer boundaries)

damercer, 2017-07-11:
That is exactly what I was thinking to myself. Robin was brave enough to 
actually say it out loud.

Doug

lclausen-adi, 2017-07-11:
The spikes are most likely an artifact of implementation not the algorithm. Changing the algorithm wont necessarily fix that.

pcercuei, 2017-07-11:
Changing the whole algorithm right now would make no sense, I'm not doing that. Again, we can fix this bug by just making sure that the buffer size of the ADC is properly calculated.

amiclaus, 2017-07-24:
On Scopy build: 3aae9de (v0.6) the bug is fixed.

amiclaus, 2017-08-16:
Scopy version: 9bf9997
It seems that the issue persists: RC LPF with Fc ~ 150Hz.
[image]

For a HPF the spikes are less pronounced:
[image]

damercer, 2017-08-16:
It looks like the bad data is happening mostly in places where the input signal level is small, -30 to -50dB. Which analog input gain range is it in when taking measurements? The high noise floor of the low gain range could be causing issues. I've noticed some of the same issues in my software as well, much worse in low gain mode but still happens a little in high gain mode when input signal level is very small.

The M2K just does not have the dynamic range of Analog Discovery or M1K.

Doug

pcercuei, 2017-08-21:
I think this is a different problem, and not a software one. Proof, increasing the amplitude of the waveform decreases the jitter.

adisuciu, 2018-01-23:
This issue seems to be solved in the lastest version of Scopy.
However there seems to be some jitter toward the far end of the frequency, which I guess makes sense. Given the fact that our sampling freq is 75MHz, we can't accurately measure the amplitude of a 23MHz signal (for example), especially if that signal has a low amplitude. My suggestion would be to limit the sweep range to 10MHz.
[image]
[image]

amiclaus, 2018-02-05:
This issue is solved on Scopy v0.10 release.

closed 2018-02-05 by amiclaus