analogdevicesinc/scopy #168: Cannot see cutoff frequency for large sweep range

amiclaus, 2017-07-10:
Scopy version: 4b0a59a, OS: Windows 7, Firmware: v0.14

Connect a simple RC HPF (Fc ~ 340Hz) to the Network Analyzer.
Make a sweep between 100Hz-10MHz.

If I want to see the cutoff frequency on the plot in order to compare it with the calculated one, I cannot do this because the step value for the cursor is too large:
[image]
The issue regarding the spikes was stated in #167 .

rgetz, 2017-07-10:
For the computer science folks - can you describe the exact circuit, showing Schematic and implimentation...

Thanks

amiclaus, 2017-07-10:
Sure: the circuit consists of a a capacitor(C) of 47nF value and resistor(R) of value 10kohm value. The circuit schematic is presented in the figure below, where V1 is the AC voltage source:
[image]
The breadboard implementation of this circuit for the network analyzer is the following:
[image]
The purpose of this circuit is to attenuate the amplitude of the signals that have a frequency less than a specific cutoff frequency that is computed using the following formula: Fc = 1/(2*pi*R*C) = 338Hz (in our case). This cutoff frequency value is considered to correspond to a -3dB attenuation of the signal (where power of the signal drops at half value/ max. amplitude voltage value is ~0.7 of the initial one). Therefore, for the sweep setup mentioned above (100Hz-10MHz) I wanted to see if the cutoff frequency (at -3dB) on the plot corresponds to the calculated value.
The circuit was taken fro the activity lab material.
[Activity, The voltage dependent capacitance of the PN junction](https://wiki.analog.com/university/courses/electronics/electronics-lab-pn-junction-cap)

amiclaus, 2017-07-24:
On Scopy build: 3aae9de (v0.6) the bug is fixed.

closed 2017-07-24 by amiclaus