analogdevicesinc/scopy #170: Signal Generator: Sine, Triangle and Sawtooth signals are distorted on odd valued frequency

drbautista, 2017-07-11:
OS: Windows 7 64bit
 Scopy Version:  95565b3
 Firmware Version: v0.14

Connect scope ch1+ to AWG ch1 and scope ch- to gnd

1. Set the signal generator's channel 1 to the following configuration:
           Waveform type: Triangle
           Amplitude: 6V - 10V  (it's easier to reproduce on higher voltage)
           Frequency: 100Hz (it's easier to reproduce on lower the 200Hz frequency)
           Offset: 0V
           Phase: 0deg
The signal is still good at this point

[image]

2. Click the knob on frequency to have an increment of only 1Hz or input 101Hz frequency. The signal will be distorted. sometimes it doesn't work on a single try but try to increment the frequency by 1 and check the oscilloscope on every odd numbered value.

[image]

The same behavior is also present on channel 2. Even in different frequency and waveform type.
[image]

[image]

pcercuei, 2017-07-17:
The signal generator switches a lot between the sample rates for the DAC to find the perfect setting.

The problem right now is that we use a single correction factor for the Volts-to-sample conversion, calculated during the calibration. But that correction factor depends on the sample rate, we probably should run the calibration at each sample rate available. I think @dNechita is working on a fix.

amiclaus, 2017-07-24:
On Scopy build: 3aae9de (v0.6) the bug is fixed.

closed 2017-07-24 by amiclaus