analogdevicesinc/scopy #171: Data logging feature

msuciu, 2017-07-12:
I think it will be a good idea to have a data logging feature for voltmeter. It will be easy to implement this by changing the orientation of the strip chart, position and probably history period to be more than 60 seconds. Also the user should have the possibility to record and export measured data, voltage in this case and temperature of the board, for long period of times, like days, if is possible.

msuciu, 2017-07-13:
Also, it will be nice to have a recording option for oscilloscope.

rgetz, 2017-07-15:
We already have an export function in the scope - but datalogging (over days/weeks), is much better suited for the volt meter. 

Data logging as nothing to do with the orientation of the strip chart - we are not changing that.

As for datalogging, it's just opening a file, appending the active current measurement (AC, DC, or RMS) of the enabled channels, and closing the file, doing a sleep() for however long, and then repeat. That way when the PC crashes, etc - the chance of the file being corrupt - is pretty low.

It's unlikely Paul will have time to do this before he leaves.

AlexandraTrifan, 2017-10-11:
For the voltmeter, this is implemented starting with commit 7c5fda48227a7872eba33da9119e481750bfaa8a .

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the enchancement is implemented.

closed 2017-11-21 by amiclaus